Sayeed97/UNO-ESP8266-I2C #6: SPI: Learn how to use SPI communication using SPI.h library

Sayeed97, 2024-05-11:
SPI.h library allows you to communicate with SPI devices, a feature on all Arduino and ESP8266 boards. SPI is a communication protocol for reading/sending data to/from devices.

References:

SPI tutorial for Arduino and ESP8266: https://diyi0t.com/spi-tutorial-for-arduino-and-esp8266/
SPI.h: https://www.arduino.cc/reference/en/language/functions/communication/spi/

charantejaabbavathini, 2024-05-12:
The above code links are for the Arduino but doesn't work for ESP8266. For Hardware Connections refer to the link https://diyi0t.com/spi-tutorial-for-arduino-and-esp8266/

ESP8266 as Slave
References:
Source code: https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/src 
Example code: https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/examples/SPISlave_Test

charantejaabbavathini, 2024-05-12:
TODO: 

Simplify the ESP8266 code to perform rudimentary functionalities of SPI slave -
1. The ability to receive the message from SPI master
2. Respond to the Master's message.

charantejaabbavathini, 2024-05-19:
TODO:
1. Learn the hspi.c source code
2. Simplify Library to make it applicable for ESP8266